ieatacid/OctoPrint-GcodeEditor #1: Cannot edit gcode files form the printer SD card

drdelaney, 2017-11-02:
Understandably, you cannot edit the gcode files from the printer SD card due to the fact that you essentially need to download it to the local device to edit, which will take time.

Though the edit button still exists, after taking a while to do anything, it shows some HTML instead of the file.

My suggestion is to see if you can limit the edit button to only show for local gcode files and not printer SD card gcode files. Though, if you can actually get it to display the SD card gcode and allow to save (even locally) that would obviously be better!

ieatacid, 2017-11-02:
Thanks for the report. I will look into it.

ieatacid, 2017-11-06:
Newest release should fix this, but let me know if you still have trouble.

closed 2017-11-10 by ieatacid