ieatacid/OctoPrint-GcodeEditor #2: Feature request: Edit "header" only

lukie80, 2017-11-18:
Hi, I think it would be useful to have an option (in plugin preferences) to edit only the first ~100 lines of gcode to speed up file transmission to the client. I assume, the typical usage scenario is probably changing temperatures, fan states and z-leveling data for which ~100 lines of gcode would be sufficient.

ieatacid, 2017-11-18:
Good idea, I'll add this.

ieatacid, 2017-11-18:
I'm not sure how to do this. I know we'd have to have the server only send X amount of lines to the client but I don't know how to go about doing that. Any ideas?

lukie80, 2017-11-18:
Thanks. Unfortunately I have hardly any knowledge of the Octoprint software architecture and also limited knowledge of JS. After reading your code, I assume you download the gcode given by the href entry of the download button into a text field and use the OctoPrint.files.upload function to send the gcode back.

There may be a way of a partial download (HTTP GET range request maybe?) if Octoprint supports it but there seems to be no "partial upload / update / overwrite" function in Octoprint.

I assume there is no way to implement the feature on client side. And even on server side I assume one has to use the provided Octoprint functions which could be quite time-consuming.

ieatacid, 2017-11-18:
These are the javascript file operations: http://docs.octoprint.org/en/master/jsclientlib/files.html

I don't think this is doable, currently.

lukie80, 2017-11-18:
I see and I agree.

ieatacid, 2017-12-14:
I think I can do this.

Or...

I can make the edit window show extruder/bed temperature, fan speed... What else would be useful information to have the ability to change?

closed 2020-12-06 by ieatacid