ieatacid/OctoPrint-GcodeEditor #4: Feature Request: Search and Replace

baxerus, 2017-11-19:
A search and replace feature would be great (extra points for RegEx 👍 , number of edits done, undo)!

ieatacid, 2017-11-20:
I'm trying to think of a good way to add this:  
* Find and Replace text input boxes in the edit window with a button to begin the search
* A 'Find/replace' button in the edit window that opens a Find/replace window
* Something else?

That's assuming I could get it to work (find/replace -- I could add the additional controls or get it to pop up another dialog box). I'm more well-versed in C/C++. I haven't done much with JS or or any web development type stuff before this >< so I'm open to ideas/suggestions/help :)

ieatacid, 2017-11-20:
Here's the first option I mentioned above. It's kind of ugly, IMO

[image]

ieatacid, 2017-12-15:
Could you check out issue #2? I know how to only send part of a file to the client interface, but I'd like to decide on one way to do it.

baxerus, 2017-12-15:
I do not know exactly how you want to find things like temps, fan speed, etc. And I do not know if this wouldn't limit the functionality of your plugin to much. But you have to decide this.

For my part, if I see "Gcode Editor" I assume, that I can edit everything (every part of the file) and not being limited to some parts and/or values.

I do not know what you are doing exactly in the moment, but isn't is possible so "simply" load the whole gcode file to the browser (with a loading animation with percentage, because this could take some time) and do the editing in JavaScript in the frontend only (Javascript can even handle RegEx)? Than uploading again and replacing the original file.

ieatacid, 2017-12-15:
> I do not know exactly how you want to find things like temps, fan speed, etc. And I do not know if this wouldn't limit the functionality of your plugin to much. But you have to decide this.

Yeah, the user would be limited to editing only the values displayed. Maybe I'll release it as 'Gcode Editor Basic' or something. The really great thing about it though, is that it can open, parse, and display the values of a 60mb file in milliseconds. That's the beauty of doing everything on the server-side of things.

> I do not know what you are doing exactly in the moment, but isn't is possible so "simply" load the whole gcode file to the browser (with a loading animation with percentage, because this could take [...]

You're correct:  the file gets downloaded to the browser where you can edit it, then when you save it, it gets uploaded back to the server. It's all done in the JS front-end.

baxerus, 2017-12-16:
I see problems, how to identify the values you want to change in the "Gcode Editor Basic". Not all Gcode files are created equal 😉 . If different Slicers are used or the user uses special custom start Gcodes or perhaps even don't use OctoPrint with a FDM-3D-Printer...

I do not know if you do yourself a favor with a Basic version. You might have many Issues, that simply result of different uses of the users.

The downloading and uploading procedure isn't the best thing you could think about, but for the (most likely rare) prupose of editing a Gcode file, I think it is okay. So, in my opinion, a more sophisticated editor would be the best.

Perhaps there is already something out there (most likely), perhaps syntax highlighting or so.

closed 2020-12-06 by ieatacid