ieatacid/OctoPrint-GcodeEditor #5: Changes are not saved

oakdesign, 2017-11-24:
Installed the GCodeEditor through Octoprint PluginManager. I can open files, and yes they are currently not loaded for print. But any changes are not saved, no error message at all. Also trying to save with a new filename doesnt do anything. Is there any logfile written to look up

ieatacid, 2017-11-24:
You can try and see if there's any info in `/home/pi/.octoprint/logs/octoprint.log`. After saving a file it writes:

```
2017-11-24 14:35:56,469 - octoprint.filemanager.analysis - INFO - Starting analysis of local:20mm-box.gcode
2017-11-24 14:35:56,469 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/devel/OctoPrint/venv/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/20mm-box.gcode
2017-11-24 14:35:58,188 - octoprint.filemanager.analysis - INFO - Analysis of entry local:20mm-box.gcode finished, needed 1.72s
```
Maybe there's an error message there in place of those lines.

If you're on Chrome you can also `ctrl+shift+i` to open the developer console and see if there is a javascript error.

oakdesign, 2017-11-24:
the log entries are written without any error, developer console shows no errors as well but still the changes are not in the file
2017-11-24 19:55:38,046 - octoprint.filemanager.analysis - INFO - Starting analysis of local:test.gcode
2017-11-24 19:55:38,074 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/test.gcode
2017-11-24 19:55:52,306 - octoprint.filemanager.analysis - INFO - Analysis of entry local:test.gcode finished, needed 14.26s

Filesize shoudn't be an issue as the testfile is only 1.2KB

ieatacid, 2017-11-24:
I'm not sure what it could be. Are you running an older version of OctoPrint?

oakdesign, 2017-11-24:
No 1.35 master-branch. Ok will see for the Moment go the old way, download edit upload. Will see to get VS Code with python up and running and do some debugging once I have some spare time

ieatacid, 2017-11-25:
That's weird: It took 14.26s to analyze a 1.2KB file. Mine shows 1.72s to analyze a 22.1KB file.

Can you send me the file (ieatacid@gmail.com) ?

wmccracken, 2017-11-29:
I'm seeing issues as well. In my case, i'm allowed to edit the file and when I save, the new file shows up in the list but shows a size of '--' . The edit icons then disappear from all of the files in the list.

If I delete the file (which sometimes requires a reboot to work properly) the edit gcode icon comes back to all files in the list.

I'm running octopi and all plugins and octoprint are up to date (as well as apt packages).

Log output (there are no errors): 
2017-11-29 00:50:39,984 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/E2Snowman_Hat_With_Spike.gcode
2017-11-29 00:50:41,158 - octoprint.filemanager.analysis - INFO - Analysis of entry local:E2Snowman_Hat_With_Spike.gcode finished, needed 1.18s

ieatacid, 2017-11-30:
> I'm seeing issues as well

Local or SD files? It only works with local files (stored on the OctoPrint server). You're both using linux, correct? (It appears so, from the log output).

Can either/both of you send me a file that fails (ieatacid@gmail.com)?  Or is it all files that fail?

I use a Pi 3 for my OP server and Linux Mint 18 for plugin dev (running its own OP server with virtual printer) and it works fine for me on both. If I can't reproduce the problem, I may not be able to fix it.

bjarchi, 2017-12-06:
I've also noticed this issue. It fails to save with any gcode file I've tried to edit. Always local files. I first noticed this behavior with the last release of the plugin, and octoprint 1.3.5, and I can confirm that it persists on the current plugin version and octoprint 1.3.6rc2 (and rc1). 

Not sure if it matters, but I'm also running octoprint on a Pi 3, installed via octopi. What can I do to help debug?

ieatacid, 2017-12-06:
Folders I can fix. I've never actually used them so that was something I overlooked.

bjarchi, 2017-12-07:
I bet that fixes the problem!

ieatacid, 2017-12-08:
Please confirm if 0.2.4 fixes this.

bjarchi, 2017-12-10:
Yes, seems to fix the problem for me as well. Thanks!!

ieatacid, 2017-12-15:
Closing this because no one else reported that the fix didn't work for them.

closed 2017-12-15 by ieatacid