ieatacid/OctoPrint-GcodeEditor #6: After saving the file is erased before printing

tgunr, 2017-12-22:
Tried using the editor to simply change the M104 setting and save the file. After save and trying to print the print finished immediately. Using the editor to look at same file again, the file was empty!

ieatacid, 2017-12-22:
What OS is the server running on (Windows, OctoPi, Ubuntu, etc.)? Is the file in question stored locally on OctoPrint or the SD card? Are there any errors in the OctoPrint or JavaScript logs? Is it a large file, perhaps?

I need more information than you provided since I've never encountered this.

tgunr, 2017-12-23:
Running Octoprint under
Linux octoprint 4.9.0-4-amd64 #1 SMP Debian 4.9.51-1 (2017-09-28) x86_64 GNU/Linux

The file was uploaded from Slic3R PE and is fine until I try to edit it. If I just look at the file then cancel it is ok. If I Save it the file becomes empty.

File is small, here is the result from ls -ltr before save:
[octoprint]/octoprint/.octoprint/uploads (582): ltr
total 18532
-rw-r--r-- 1 octoprint octoprint 2021942 Dec 17 17:10 Base-2.gcode
-rw-r--r-- 1 octoprint octoprint 4785213 Dec 19 09:53 BaseFinal.gcode
-rw-r--r-- 1 octoprint octoprint  268060 Dec 20 02:08 Basetest.gcode
-rw-r--r-- 1 octoprint octoprint  244965 Dec 20 02:37 Basetest-1.gcode
-rw-r--r-- 1 octoprint octoprint 9916252 Dec 20 10:05 lithoBase-180mm-1.gcode
-rw-r--r-- 1 octoprint octoprint       0 Dec 22 08:48 extruder-body.gcode
-rw-r--r-- 1 octoprint octoprint 1732143 Dec 22 10:14 FanCam-1.gcode

After save
[octoprint]/octoprint/.octoprint/uploads (582): ltr
total 16840
-rw-r--r-- 1 octoprint octoprint 2021942 Dec 17 17:10 Base-2.gcode
-rw-r--r-- 1 octoprint octoprint 4785213 Dec 19 09:53 BaseFinal.gcode
-rw-r--r-- 1 octoprint octoprint  268060 Dec 20 02:08 Basetest.gcode
-rw-r--r-- 1 octoprint octoprint  244965 Dec 20 02:37 Basetest-1.gcode
-rw-r--r-- 1 octoprint octoprint 9916252 Dec 20 10:05 lithoBase-180mm-1.gcode
-rw-r--r-- 1 octoprint octoprint       0 Dec 22 08:48 extruder-body.gcode
-rw-r--r-- 1 octoprint octoprint       0 Dec 22 10:26 FanCam-1.gcode

As you can see, it  got truncated.

Nothing of interest in the log

2017-12-22 10:14:41,093 - octoprint.filemanager.analysis - INFO - Analysis of entry local:FanCam-1.gcode finished, needed 10.15s
2017-12-22 10:14:43,867 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-12-22 10:15:55,034 - octoprint.plugins.filamentmanager - INFO - Filament used: 123.7703 mm (tool0)
2017-12-22 10:15:56,979 - octoprint.plugins.nautilus - INFO - >>>> Sending notification type [error] to 0 device(s)
2017-12-22 10:19:24,831 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-12-22 10:20:47,017 - octoprint.plugins.filamentmanager - INFO - Filament used: 139.65513 mm (tool0)
2017-12-22 10:20:49,087 - octoprint.plugins.nautilus - INFO - >>>> Sending notification type [error] to 0 device(s)
2017-12-22 10:22:22,551 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-12-22 10:25:12,834 - octoprint.plugins.filamentmanager - INFO - Filament used: 209.51393 mm (tool0)
2017-12-22 10:25:13,723 - octoprint.plugins.nautilus - INFO - >>>> Sending notification type [error] to 0 device(s)
2017-12-22 10:26:41,661 - octoprint.filemanager.analysis - INFO - Starting analysis of local:FanCam-1.gcode
2017-12-22 10:26:41,678 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /octoprint/OctoPrint/venv/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /octoprint/.octoprint/uploads/FanCam-1.gcode
2017-12-22 10:26:44,393 - octoprint.filemanager.analysis - INFO - Analysis of entry local:FanCam-1.gcode finished, needed 2.73s

ieatacid, 2017-12-23:
Can you email me the file?

ieatacid@gmail.com

tgunr, 2017-12-23:
Which file? The gcode? All gcode files act the same.

tgunr, 2017-12-23:
I can share screen if you like to nail this down.

ieatacid, 2017-12-24:
Have you installed the 0.2.5 update? If the problem persists after updating we'll have to make plans to do that.

tgunr, 2017-12-24:
Yes, updated.

ieatacid, 2017-12-24:
> Yes, updated.

And, still, the same problem?

tgunr, 2017-12-24:
Correct, I have the octoprint running as a container under proxmox, can give you access if you want to poke around

tgunr, 2017-12-24:
Send me your ssh public key and I will add it to the root then you can ssh into octoprint.polymicro.net <http://octoprint.polymicro.net/>.

ieatacid, 2017-12-24:
I sent it via email to the address in your profile.

tgunr, 2017-12-24:
ok, you should be able to get in now, are you on Mattermost or Slack?

I am on MatterMost at 3d printing, send me your email address. Im davec@polymicro.net

ieatacid, 2017-12-24:
ieatacid@gmail.com

tgunr, 2017-12-24:
I am tgunr on general channel, you can PM me there

> On Dec 24, 2017, at 8:55 AM, Jason F <notifications@github.com> wrote:
> 
> ieatacid@gmail.com <mailto:ieatacid@gmail.com>

ieatacid, 2017-12-24:
I have slack. I looked at MatterMost but it says I have to install database and proxy software.

masto, 2018-01-13:
I'm having this same issue, I think. When I edit a file and save it, it ends up zero bytes. Doesn't seem to matter what I do or whether I enter a new name. I'm going to try to debug it today but wanted to check if this was a known issue.

GcodeEditor (0.2.5) on OctoPrint 1.3.6 on a Raspberry Pi (OctoPi)

masto, 2018-01-14:
So in my case, this was a client issue. Safari has some issue with the File API, but fortunately the workaround is quite simple. I just sent a pull request for c3175c5aa2ef8e75759e249f9b0cff28a548752c which fixes the problem for me. Not sure if it's the same cause of what @tgunr was experiencing.

ieatacid, 2018-01-15:
If 0.2.6 fixes this for anyone still having problems, please let me know so I can close this.

ieatacid, 2018-01-20:
tgunr reported to me that this fixed his issue so I'm closing this.

closed 2018-01-20 by ieatacid