ieatacid/OctoPrint-GcodeEditor #8: Poor performance on most recent Chrome and Edge browsers

Zwaubel, 2018-08-17:
Hey,

thank you for this plugin, it's nice to be able to edit gcode files in OctoPrint!
However the performance of this plrugin in on most recent Chrome and Edge browsers is really bad.
In Chrome I'm unable edit gcode files: After opening a file and for instance marking a section and hitting the delete key, the GUI freezes (not actual freezing and not responding like a program error but just not repsonding to any input). In Edge the performance is a little better, but depending heavily on the size of the gcode file. Big files tend to make the UI freeze, too.
I'm running the latest OctoPrint on a RasPi 3 B+, so there should be planty of power. 

Any tips and tricks that I could perform to improve the performance?
I attached a OctoPrint log which shows a error with my printer stats plugin, everything else seems fine to me.

[octoprint.log](https://github.com/ieatacid/OctoPrint-GcodeEditor/files/2296764/octoprint.log)

ieatacid, 2018-08-18:
Can you send me one of these large files that aren't working correctly?

Google Drive, Dropbox, email (ieatacid@gmail.com), or whatever works for you.

Zwaubel, 2018-08-18:
Sure, here you go:

[gcode.zip](https://github.com/ieatacid/OctoPrint-GcodeEditor/files/2299585/gcode.zip)

ieatacid, 2018-08-18:
Unfortunately I don't think there's anything I can do about it. I'm running it in a VM at the moment since I'm at work and, while it's very slow, I can't get it to freeze. As you can see in the image below, Chrome is eating up a lot of RAM just to render all that text.

[image]

Zwaubel, 2018-08-22:
Fair enough, I guess then my computer is the bottleneck (it's a pretty weak one).
Closing this issue for now.

closed 2018-08-22 by Zwaubel