ieatacid/OctoPrint-GcodeEditor #13: Very Slow

patrick505, 2020-04-05:
I want to love this plugin but it functions so slow that it makes it unusable. Whenever I click the edit button, it may take up to 20 seconds for the editor window to open. Then, when editing text, every action takes 10-20 seconds to complete. That includes every click, keyboard input, etc. Many times it freezes all together and I have to refresh the page to close the window. How can I fix it? Of course, it could be my system but since I don't have any other performance issues within octoprint or anywhere else on the computer, I don't think that is it.
Running on 2015 Imac, using Chrome.
Thanks for any help

a8ree, 2020-05-02:
Yep, I see the same issue.

Also running Chrome on OSX

peco8645, 2021-03-19:
Same here, chrome just freezes. Windows 10, latest chrome 64 build

closed 2023-04-18 by ieatacid