ieatacid/OctoPrint-GcodeEditor #18: Plugin Breaks File Upload When Adding Button

jneilliii, 2020-08-05:
Due to recent changes in OctoPrint 1.4.1 and the file list, this plugin is breaking some things. Your add button method is causing file uploads to get stuck. The line the error happens on is [here](https://github.com/ieatacid/OctoPrint-GcodeEditor/blob/002843ea6873f53f63e3c44db5c3473fcf2e5ec1/octoprint_GcodeEditor/static/js/GcodeEditor.js#L143). The error is shown below.

[image]

Relative issue ticket on OctoPrint is [here](https://github.com/OctoPrint/OctoPrint/issues/3675).

jjjdor, 2020-08-06:
I can't even see the edit button.

jneilliii, 2020-08-06:
It's because it's not getting added. Relative to how @ieatacid is adding the button on the javascript side.