ieatacid/OctoPrint-GcodeEditor #20: No update if Octoprint is in virtual environment.

xeddog, 2020-08-28:
I'm not sure if this is the correct place for this, but here goes.

I have three Raspberry Pi's, each attached to it's own printer.  One of these received an update to gcode editor 0.2.9 which installed and appears to be working.  The other two have not, even using the force check, and it has been a few days now.  On the two that did not get the update, the plugin is still shows as  0.2.8, and is blacklisted.  

The significant difference between these three Pi's seems to be that the one that DID get the update was an Octopi install, where the other two were Raspbian Buster installations that then had Octoprint installed into a virtual environment.

cp2004, 2020-08-29:
OctoPi has OctoPrint installed in a virtual environment too.

xeddog, 2020-08-29:
> 
> 
> OctoPi has OctoPrint installed in a virtual environment too.

Not this one.  I don't remember what version of OctoPi it was, but it was before venv came up from the depths of hell.

lienbacher, 2020-09-10:
Whatever the issue is, I am also not able to update on an octopi install. only shows 0.2.8 as blacklisted.

cp2004, 2020-09-10:
@lienbacher You need to reinstall manually, since it is blacklisted. There is a reinstall button in the plugin manager.

xeddog, 2020-09-10:
I'm sure that I posted a closing for this issue, but I guess something else happened.  Anyway, when I was re-imaging all three of my RPi's, I noticed a setting that might explain everything.  In the Settings>Server at the bottom, there is a setting that allows you to enable/disable blacklist processing.  I'm guessing that it was disabled on the two of them that did not update.  I have no idea why, but it would explain it all.

closed 2020-09-10 by xeddog