ieatacid/OctoPrint-GcodeEditor #23: Version 0.2.9 not compatible with Octoprint 1.5.0

Boerner1337, 2020-12-03:
Hey there, 

first of all: great job you did here with the plugin, I am using it on a daily basis. 
After upgrading to Octoprint 1.5.0 it seems that Version 0.2.9 of GCode Editor is not compatible anymore. 
At least I cannot see the "edit" button next to my gcode files anymore, if you find the time to fix that, I'd really appreciate it as I am using your plugin quite recently. 
That's not super suprising as these things usually happen when upgrading to a super new Octoprint version but nevertheless I just want to leave the information here for you :)

Thanks, 
Kevin

CmdrCody51, 2020-12-03:
I second this. I think this is one dam good plugin when it works. I have also used this plugin every day.
Same problem - no edit button with 1.5.0
And for Issue #4 - can't you just use the browser find-on-page?
Keep up the good work!

sirtuxalot, 2020-12-04:
Confirmed on v1.5.1 as well.  Looking forward to next update as it is a very handy tool.  Well Done!

Whitehawk2000, 2020-12-05:
I can confirm it. 
Keep up the good work!

cmoore159, 2020-12-05:
Confirmed same problem on v1.5.1 as well. Looking forward to next update as it is a very handy tool. Well Done!

calexandre, 2020-12-05:
+1 also broken for me.

ieatacid, 2020-12-06:
Should be fixed now

CmdrCody51, 2020-12-06:
Works for me.  And so does the browser find in page - Woot!

sirtuxalot, 2020-12-06:
Thank you sir, you are a gentleman and a scholar!

Whitehawk2000, 2020-12-06:
Great, after a reinstall of the plugin everything is working fine! Thank you!

cmoore159, 2020-12-06:
Thank you very much after a reinstall of the plugin everything is working fine on 1.5.1! Keep up the awesome work.

closed 2020-12-06 by ieatacid

Boerner1337, 2020-12-06:
Thank you very much! All works well now!