ieatacid/OctoPrint-GcodeEditor #25: Editor replacing space by underscore on file save

clementprevot, 2021-02-10:
Hi!

First of all, thanks a lot for this plugin, it works perfectly and is really useful.
There is only one thing that is "bothering" me: when I edit a file whose name contain a space, upon save the file is renamed and all the spaces are replaced by underscore.

As I can see from the code, it's coming from [the sanitize function](https://github.com/ieatacid/OctoPrint-GcodeEditor/blob/master/octoprint_GcodeEditor/static/js/GcodeEditor.js#L305) called [during the save](https://github.com/ieatacid/OctoPrint-GcodeEditor/blob/master/octoprint_GcodeEditor/static/js/GcodeEditor.js#L28)

I don't see any issue with file with spaces in their name (when I upload files with spaces they work perfectly fine and if I also rename them with space using File Manager, no issue either) so I was wondering why you are stripping spaces?

If there is no particular reason, I'll be glad to submit a PR to change this behavior 🙂

ieatacid, 2021-02-11:
I was either following OctoPrint's behavior or it was something from another plugin. If it works without it then, by all means, submit a PR and I'll merge it 🙂

ieatacid, 2021-02-11:
Does it work if you just remove the call to sanitize?
i.e.
```
- var fName = self._sanitize(self.destinationFilename());
+ var fName = self.destinationFilename();
```

clementprevot, 2021-02-11:
Well, yes it would work but it would definitely cause issues because this sanitize function is not only cleaning spaces but also all "weird" character.
So keeping the spaces is safe, but keeping other exotic character may not be.
It can be worth checking it, but I won't have time for that for the moment.

So I suggest just doing the space thing and see later for the rest of the characters.

Does it works for you?

ieatacid, 2021-02-11:
Yes, that's good.

PR #26 (2021-02-11): fix(sanitize): allow spaces in the filename
It looks like Octoprint doesn't have any issue for handling filenames with spaces. So there is no need to sanitize them.

Fixes #25
closed 2021-02-11 by ieatacid