ieatacid/OctoPrint-GcodeEditor #27: [Bug] after latest update, "Update" reminder continues to appear, even after updating

KenLucke, 2021-02-11:
Version showing in Plugin manager is 0.2.10, update supposedly bumps to 0.2.11.  Did you forget to change the version string, or is your plugin just lonely? :)

ieatacid, 2021-02-11:
I just merged a pull request that fixed a bug and created a release.

KenLucke, 2021-02-11:
Well, it's still showing 0.2.10 in the plugin manager and bugging about needing an update to 0.2.11 every time OP is restarted, even after a "Force Check" for plugin status.

[image]

jneilliii, 2021-02-12:
You need to update the version number in setup.py @ieatacid, and re-create the release, or bump the version to 0.2.12 and create a new release that matches that version.

https://github.com/ieatacid/OctoPrint-GcodeEditor/blob/62fd35c6a35021261b02835e09c5db7e39172b47/setup.py#L17

HollyRidge, 2021-02-12:
Yes, I am getting the same thing too. I have run the update now several times and is stuck in that loop. The plugin manager also shows 0.2.10 version installed as well. This is the output I get when I re-run the update again in case that helps...

`
Update successful, restarting!
The update finished successfully and the server will now be restarted.
Updating...
Updating, please wait.
+++++++++++++++++++++++++++++++++++++++++
Now updating GcodeEditor Plugin to 0.2.11
+++++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/ieatacid/OctoPrint-GcodeEditor/archive/0.2.11.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/ieatacid/OctoPrint-GcodeEditor/archive/0.2.11.zip
Downloading https://github.com/ieatacid/OctoPrint-GcodeEditor/archive/0.2.11.zip (122 kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-GcodeEditor==0.2.10) (1.5.3)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2020.11.13)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (5.3.1)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (7.1.2)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.57.0)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2.11.2)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2.0)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (4.0)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.1.1)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.1.2)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.10.4)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.12.1)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.4.21)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.0.0)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (3.1.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.1.5.post0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.4)
Requirement already satisfied: zeroconf<0.25,>=0.24 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.24.5)
Requirement already satisfied: feedparser<7,>=6.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (6.0.2)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.1.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2.8.5)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.10.9)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.2)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.1.1)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2021.2.4)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.2.0)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2.25.1)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.6.0)
[... cut ...]
Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoPrint-GcodeEditor==0.2.10) (2.10)
Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->OctoPrint-GcodeEditor==0.2.10) (1.15.0)
Requirement already satisfied: ifaddr in /home/pi/oprint/lib/python3.7/site-packages (from zeroconf<0.25,>=0.24->OctoPrint->OctoPrint-GcodeEditor==0.2.10) (0.1.7)
The update finished successfully and the server will now be restarted.`

ieatacid, 2021-02-12:
@jneilliii I'm aware of that, obviously 🙄

Will-wastelander, 2021-02-12:
Watching

Daskew78, 2021-02-13:
same here on 2 different installations. Even tried uninstalling and re-installing.

Will-wastelander, 2021-02-13:
I have 2 systems that I have this software installed. Awaiting new release.

Hobby4life, 2021-02-13:
You can easily fix it your self if you wish.

Download the repository, unpack it.. edit setup.py and change  plugin_version = "0.2.10" to  plugin_version = "0.2.11" 

Repack the whole thing into a zip file.. and manually install it.

[image]

Hypertoken, 2021-02-13:
> You can easily fix it your self if you wish. ...

Thanks this works.

Will-wastelander, 2021-02-13:
I tried this last night and it didn't work for me. What lines are you
updating, and in what files ? I updated setup.py.

Hobby4life, 2021-02-13:
> I tried this last night and it didn't work for me. What lines are you updating, and in what files ? I updated setup.py.
> […](#)

Try updating with this url: https://github.com/Hobby4life/OctoPrint-GcodeEditor/archive/master.zip

closed 2021-02-13 by ieatacid