ieatacid/OctoPrint-GcodeEditor #28: Nothing happens after save

Mesmer88, 2021-06-02:
I notice that with OctoPrint 1.6.1 the plugin not longer works.
After clicking save the popup window closes but the file is not edited.

In browser decveloper mode i see the POST message gets a 415 UNSUPPORTED MEDIA TYPE error.

Request header:

```
Content-Disposition: form-data; name="file"; filename="Calibration/CE5_5cm_Square.gcode"
Content-Type: text/plain

```

This is re response to the request:

`{"error":"Can not upload file, wrong format?"}`

Loewe1000, 2021-06-23:
Getting the same error

StoneAgeSkillz, 2021-07-09:
i have the same issue, i think it could be perrmission related, or username related, because i changed mine

anthonyonions, 2021-08-17:
I've just updated to OctoPrint 1.6.1and have the same problem.

@ieatacid - any time scales on when this will be fixed? I run 2 printers and editing the gcode to suit has been a life saver til now.

sergeysi779, 2021-09-30:
I've ran into similar issue. For me editing works in root folder but not in subfolders.
To fix the issue you have to change line 33 in static/js/GcodeEditor.js from
`OctoPrint.files.upload("local", file, { filename: _selectedFilePath + fName });`
to
`OctoPrint.files.upload("local", file, { filename: fName, path: _selectedFilePath });`

Full path for the file should be something like:
`/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_GcodeEditor/static/js/GcodeEditor.js`

Mesmer88, 2021-10-02:
@sergeysi779 

Thanks a lot, it fixed it for me as well!
One mention though, on my setup the path was different:

`/home/pi/oprint/lib/python3.7/site-packages/octoprint_GcodeEditor/static/js`

anthonyonions, 2021-10-02:
I tried this yesterday but didn't seem to make a change on either of my installs. 
Does anything need doing other than a restart?

Mesmer88, 2021-10-02:
I just edited the line and restarted Octoprint from the menu. Maybe double check the edit?

anthonyonions, 2021-10-02:
Thanks for confirming. Appears my js skills are lacking - I'd used the wrong comment format to comment out the original line of code. All sorted now, thanks @Mesmer88  and thanks @sergeysi779 for finding the fix originally.

FindusD, 2021-12-12:
I have the same problem with OctoPrint 1.7.2:
I've found the file GcodeEditor.js in 
/home/pi/oprint/lib/python3.7/site-packages/octoprint_GcodeEditor/static/js
I have made the changes described and restarted OctoPrint.
Now, when I click on "Save", the editor window does not even close. And it is not saved either.

fleinze, 2022-01-02:
I can confirm this is working with Octoprint 1.7.2.

vulgerrity, 2022-04-10:
This worked for me in 1.7.3!

Build-0-Matic, 2022-05-18:
Still not working on Octoprint 0.18
Tried editing the line in the GcodeEditor.js that is located here
./oprint/lib/python3.7/site-packages/octoprint_GcodeEditor/static/js/

I rebooted the Octoprint server but it still doesn't work.

Fsie, 2022-06-03:
In 1.8.1 the told Code is existing twice. Changing both as adviced Boy sergeysi779 does make the trick. Runs like a charme.

Build-0-Matic, 2022-06-03:
This has got to be the strangest bug as it now works fine...
I just don't get it but it seems OK. Slow but OK.

ieatacid, 2022-06-03:
If someone submits a pull request I'll merge it.

closed 2023-04-18 by ieatacid