ieatacid/OctoPrint-GcodeEditor #31: Changes are not saved Returns

vulgerrity, 2022-04-10:
The issue from Issue #5 seems to have returned.  I cannot save Gcode to files that are in sub directories, but I can edit GCode files in the root directory only.  When trying to save in a subdirectory the console reports a Post error 415 (UNSUPPORTED MEDIA TYPE).

vulgerrity, 2022-04-10:
I was able to confirm that the solution [here](https://github.com/ieatacid/OctoPrint-GcodeEditor/issues/28) works in OctoPrint 1.7.3

Build-0-Matic, 2022-05-18:
But it sadly stopped working 0n version 1.8.0

coelholm, 2023-01-27:
Check on that https://github.com/ieatacid/OctoPrint-GcodeEditor/issues/33#issuecomment-1407159709

TylonHH, 2023-01-31:
works for me: https://github.com/ieatacid/OctoPrint-GcodeEditor/releases/tag/0.2.13

closed 2023-04-17 by ieatacid

7s3, 2024-03-01:
I'm also having this problem in GcodeEditor 0.2.14