ieatacid/OctoPrint-GcodeEditor #33: cant save because name is wrong red

TylonHH, 2022-06-03:
Hey
as you can see in the picture, the name of the file is wrong. I have to correct the filename to save.
[image]
Im on OctoPrint Version 1.8.1
cheers

coelholm, 2023-01-27:
I may have find the solution, did not extensively tested, but it is working. Pull request #37

coelholm, 2023-01-27:
Filename is retrieved with command:
`var fname = button.parent().parent().children('div.title').html();`

I believe that div titles got an update from octoprint or another plugin. That update included an "<i class ...." statement creating an issue. Changing from .html() to .text() solved the problem. I've created a pull request #37, if you can test it and post your comments would be great.
You can also try to update ~/oprint/lib/python3.7/site-packages/octoprint_GcodeEditor/static/js/GcodeEditor.js and make that line looks like bellow:
`var fname = button.parent().parent().children('div.title').text();`. 

Don't forget to restart octoprint.

TylonHH, 2023-01-31:
https://github.com/ieatacid/OctoPrint-GcodeEditor/releases/tag/0.2.13

closed 2023-01-31 by TylonHH